Kinect/PyKinect2 #17: asset fails on sizeof == 72

drglitch, 2015-04-13:
Actual sizeof reported is 80. please investigate

josefmonje, 2015-05-14:
```
----> 1 from pykinect2 import PyKinectV2

C:\Python27\lib\site-packages\pykinect2\PyKinectV2.py in <module>()
   2214     ('reserved', c_ulong),
   2215 ]
-> 2216 assert sizeof(tagSTATSTG) == 72, sizeof(tagSTATSTG)
   2217 assert alignment(tagSTATSTG) == 8, alignment(tagSTATSTG)
   2218 IAudioBeamList._methods_ = [

AssertionError: 80
```

luochenhuan, 2015-12-15:
@drglitch I had the same issue. Then I uninstalled 64-bit version and installed the 32-bit Anaconda. It solves the problem, though I don't know why we cannot use 64-bit Anaconda. It also includes NumPy.

SSchwoebel, 2016-07-10:
I have the same issue. I have a 64-Bit Anaconda installed, and I tried to set up a separate 32-Bit environment with it, but the same assertion error keeps popping up. I can not uninstall my Anaconda, because I have it set up for other projects needing 64-Bit. Can someone point me to a solution that does not require uninstalling my set up?

closed 2016-09-28 by vladkol

AugsEU, 2025-09-15:
This issue still occurs when installing via pip on a 64-bit windows machine.

I have commented the sizes of each field along with a running total assuming 8 byte allignment.

tagSTATSTG._fields_ = [
    ('pwcsName', WSTRING),         #8 Total:0
    ('type', c_ulong),                       #4 Total:8
    ('cbSize', _ULARGE_INTEGER),  #8 Total:16
    ('mtime', _FILETIME),                 #8 Total:24
    ('ctime', _FILETIME),                  #8 Total:32
    ('atime', _FILETIME),                  #8 Total:40
    ('grfMode', c_ulong),                #4 Total:48
    ('grfLocksSupported', c_ulong),#4 Total: 52
    ('clsid', GUID),                           #16 Total:56               
    ('grfStateBits', c_ulong),            #4 Total:72
    ('reserved', c_ulong),                 #4 Total:76
]                                                          Total:80

So this is where the 80 figure comes from. The only explanation is that WSTRING is 4 bytes on 32-bit machines?