Kinect/PyKinect2 #22: CoordinateMapper : wrong profile for MapDepthFrameToColorSpace

pgrassia-opri, 2015-06-04:
Using the following snippet, where mapper is a coordinatermapper object result in a "python has stopped working" error message
'''		
CSP_Count=kinect._depth_frame_data_capacity 
CSP_type= _ColorSpacePoint * CSP_Count.value
CSP=ctypes.cast(CSP_type(), ctypes.POINTER(_ColorSpacePoint))
mapper.MapDepthFrameToColorSpace(kinect._depth_frame_data_capacity,kinect._depth_frame_data, CSP_Count, CSP)
'''

that snippet is conform to the profile of the function defined in PyKinectV2.py which in turn is conform to the documentation [https://msdn.microsoft.com/en-us/library/windowspreview.kinect.coordinatemapper.mapdepthframetocolorspace.aspx]

I got the function to perform properly by changing the profile of the method to use an array of c_float instead of _ColorSpacePoint  (PyKinectV2.py line 2114) and changing the cast accordingingly.

I'm using Python 2.7.9 32 bit (one of the libs I use does not perform well in 64 bit) and the lasted kinect v2 sdk (1409)

abhirajD, 2016-03-13:
@pgrassia-opri how did you access the c_float array through pyKinectRuntime

sarabsethi, 2018-04-25:
@pgrassia-opri sorry, I'm quite unfamiliar with ctypes - how do you retrieve the mapped data after running the above code? Thanks