Kinect/PyKinect2 #24: No documentation available to learn about the various functions and the parameters required for them.

DemonicIllusion, 2015-10-09:
i am a newbie to pykinect2, and there are no reliable sources available to get to know about the functions included in the library. also the example included in the library, doesn't help much and there is no help available when we want to utilise the Depth and IR stream. Moreover, there also are significant changes in the library when compared to pykinect, and hence the examples available for it dont help either. 
i would be grateful for any help!

chriss2401, 2016-08-24:
+1. Please any help understading how to connect to Kinect's depth sensor ?

nitzel, 2017-04-04:
+1

@chriss2401 Take the Infrared Example and replace `infrared` with `depth`. Do it case sensitive.
Start with `self._kinect = PyKinectRuntime.PyKinectRuntime(PyKinectV2.FrameSourceTypes_Depth)`  ...

I'd like to have documentation, too. I managed to convert the depth image to a 3D Pointcloud (with the mapper), but now I have a memory leak I can't pinpoint/fix.

```
        depth_frame = kinect.get_last_depth_frame()
        # size must be what the kinect offers as depth frame
        assert L == depth_frame.size
        # create C-style Type
        TYPE_CameraSpacePoint_Array = PyKinectV2._CameraSpacePoint * L
        # instance of that type
        csps = TYPE_CameraSpacePoint_Array()
        # cstyle pointer to our 1d depth_frame data
        ptr_depth = np.ctypeslib.as_ctypes(depth_frame.flatten())
        # calculate cameraspace coordinates
        error_state = self._kinect._mapper.MapDepthFrameToCameraSpace(
                                                                L, ptr_depth,
                                                                L, csps)
        # 0 means everythings ok, otherwise failed!
        if error_state:
            raise "Could not map depth frame to camera space! "
            + str(error_state)

        # convert back from ctype to numpy.ndarray
        pf_csps = ctypes.cast(csps, ctypes.POINTER(ctypes.c_float))
        data = np.copy(np.ctypeslib.as_array(pf_csps, shape=(self.HEIGHT, self.WIDTH,
                                                     3)))
        del pf_csps, csps, ptr_depth, TYPE_CameraSpacePoint_Array
        return data
```

Ddylen, 2020-01-29:
I have included some code on how to save colour and depth data from the kinect, and how to extract 3D positions from the saved data. Note this code is all offline (i.e. save data, then find 3D positions from it). Im not claiming this code is great, and it could probably be improved, but its better example code than I could find anywhere else
```
import numpy as np
import cv2
import pickle
import time 
import datetime

from pykinect2 import PyKinectV2
from pykinect2.PyKinectV2 import *
from pykinect2 import PyKinectRuntime

def save_frames(FILE_NAME):
    #records and saves colour and depth frames from the Kinect
    
    print("Saving colour and depth frames")
    
    # define file names
    depthfilename = "DEPTH." + FILE_NAME +".pickle"
    colourfilename = "COLOUR." + FILE_NAME +".pickle"
    depthfile = open(depthfilename, 'wb')
    colourfile = open(colourfilename, 'wb')
    
    #initialise kinect recording, and some time variables for tracking the framerate of the recordings
    kinect = PyKinectRuntime.PyKinectRuntime(PyKinectV2.FrameSourceTypes_Color | PyKinectV2.FrameSourceTypes_Depth)
    starttime = time.time()
    oldtime = 0
    i = 0
    fpsmax = 0
    fpsmin = 100
    
    display_type = "COLOUR"
    #display_type = "DEPTH"
    
    # Actual recording loop, exit by pressing escape to close the pop-up window
    while True:
        
        if kinect.has_new_depth_frame() and kinect.has_new_color_frame() :
            elapsedtime = time.time()- starttime
            if(elapsedtime> i/10):
                
                #Only for high i try evalutaing FPS or else you get some divide by 0 errors
                if i >10:
                    try:
                        fps =  1/(elapsedtime - oldtime)
                        print(fps)
                        if fps> fpsmax:
                            fpsmax= fps
                        if fps < fpsmin:
                            fpsmin = fps
    
                    except ZeroDivisionError:
                        print("Divide by zero error")
                        pass
                    
                oldtime = elapsedtime
                
                #read kinect colour and depth data (somehow the two formats below differ, think one is and one isnt ctypes)
                depthframe = kinect.get_last_depth_frame() #data for display
                depthframeD = kinect._depth_frame_data
                colourframe = kinect.get_last_color_frame()
                colourframeD = kinect._color_frame_data
                
                #convert depth frame from ctypes to an array so that I can save it
                depthframesaveformat = np.copy(np.ctypeslib.as_array(depthframeD, shape=(kinect._depth_frame_data_capacity.value,))) # TODO FIgure out how to solve intermittent up to 3cm differences
                pickle.dump(depthframesaveformat, depthfile)
                
                #reformat the other depth frame format for it to be displayed on screen
                depthframe = depthframe.astype(np.uint8)
                depthframe = np.reshape(depthframe, (424, 512))
[... 186 lines cut ...]
    out.write(frame)
    cv2.imshow('Stiching Video',frame)

    #Cycle through the rest of the colour frames, stiching them together
    while True:
        try:
            frame = pickle.load(datafile)
            out.write(frame)
            cv2.imshow('Stiching Video',frame)
            if (cv2.waitKey(1) & 0xFF) == ord('q'): # Hit `q` to exit
                break
        except EOFError:
            print("Video Stiching Finished")
            break

    # Release everything if job is finished
    out.release()
    cv2.destroyAllWindows()
    
if __name__ == '__main__': 
    
    #replace name below with the corresponding section of the name of your saved depth data (for reference, the full name of my saved colour data file was COLOUR.test.1.29.13.17.pickle)
    frames_to_video('test.29.1.13.23')

```