Kinect/PyKinect2 #26: z coordinates for skeleton

guillaumevi, 2015-12-15:
Hi, Is it possible to retrieve the z coordinates for skeleton ? I'm trying by inspriring of the exemple, but ColorSpacePoint don't have it, and if i add it myself i don't retrieve it in the joints later ...

vladkol, 2015-12-15:
Z only exists in depth space. You may use _mapper.MapCameraPointToDepthSpace from low-level API for mapping body joints to depth space. Or you can use body_joint_to_color_space and body_joints_to_color_space I've just added.

karansaxena, 2016-08-08:
@vladkol 

joints = body.joints
                    # convert joint coordinates to color space 
                    joint_points = self._kinect.body_joints_to_depth_space(joints)
                    print joint_points[PyKinectV2.JointType_AnkleLeft].x
                    print joint_points[PyKinectV2.JointType_AnkleLeft].y
                    print joint_points[PyKinectV2.JointType_AnkleLeft].z    ###This gives error.

vladkol, 2016-08-08:
You need to get z from depth frame using x and y you got from body_joints_to_depth_space

Ugur-Ayvaz, 2017-03-09:
Hi Vlad,
Thanks for your comment. Can we use the following equation to calculate z from x and y  ? Or any other equation ?

z = sqrt(x² + y² - 2 * x * y * cos(γ))

Ugur-Ayvaz, 2017-03-09:
@vladkol

vladkol, 2017-03-09:
I don't think so, unless your Kinect is facing a surface that may be defined by an equation. Depth frame has Z-data, you can get it from there.

closed 2017-03-09 by vladkol