Kinect/PyKinect2 #27: track only ONE body

LabIO, 2016-03-10:
Hello

I am trying to change the example PyKinectBodyGame.py to track only one person at a time. 

I set to 1 the variable KINECT_MAX_BODY_COUNT in PyKinectRuntime object but it does not limit the number of bodies to 1. 
I also limited the for loop in line 146 to 0, but the id number of the detected body does not start at 0 (I can see that there are detected bodies but none is drawn).

Is the body ID number stored in the camera? And if so, how could I solve this? I basically need to have only one player at a time without being disctracted by other persons entering the scene.

Thanks for your time, any feedback is appreciated.

Mick

I am a beginner in python: I hope somebody can help

FarshidAhmadi, 2017-06-01:
hi to all
it was exactly my question 
Except that members  I am a beginner in Vb.net

nitzel, 2017-06-01:
Haven't tried this, just a guess:
```
trackedBodyId = -1;
...
when a new body appears:
  if trackedBodyId==-1:
    trackedBodyId = new body id
...
CheckBody(trackedBodyId) # but ignore others
```
So while the Kinect SDK is constantly tracking all available bodies, you only look at the values of one of them.

FarshidAhmadi, 2017-08-09:
my problem solved by

   Using frame = reference.BodyFrameReference.AcquireFrame()

     bf = frame
                If bf IsNot Nothing Then
                    canvas.Children.Clear()

                    Me._bodies = New Body(bf.BodyFrameSource.BodyCount - 1) {}

                    bf.GetAndRefreshBodyData(Me._bodies)
                    dataReceived = True
                End If
                If dataReceived Then
                    Dim body As Body = Nothing
                    If Me.bodyTracked Then
                        If Me._bodies(Me.bodyIndex).IsTracked Then
                            body = Me._bodies(Me.bodyIndex)
                        Else
                            bodyTracked = False
                        End If
                    End If

                    If Not bodyTracked Then

                        For i As Integer = 0 To Me._bodies.Count - 1
                            If Me._bodies(i).IsTracked Then
                                Me.bodyIndex = i
                                Me.bodyTracked = True
                                Exit For
                            End If
                        Next i
                    End If

                    If body IsNot Nothing AndAlso Me.bodyTracked AndAlso body.IsTracked Then

TheSOMGuy, 2017-08-09:
Hi,

with the solution you wrote you can't be sure that you constantly track the same body.
Sometimes bodies disappear and reappear in the same run.
The SDK tries to reassign the same Skeleton-ID but this isn't always possible.
So the tracked ID could change in the run.
Another fact is that the same body isn't always the first ( or n-th ) entry in the queue.
Your solution tracks always one body, but i guess it's not always the same.

Cheers

nitzel, 2017-08-09:
Hi,

to handle this problem, you could try comparing rather static body metrics like ankle to knee distance (=shin length) etc. 
But with different subjects that have similar body metrics you'll still have the same problem, so I guess it is best to state some requirements like: Body always has to stay in view.

Best regards.

FarshidAhmadi, 2017-08-14:
Hi,
exactly i am not sure about trackID because every time is changing so i need unique track id for every person in bodyframe 
therefore i have to wait for corrrected SDK that solves many problem in kinect v2 like above problem and pause method for kinect sensor

many thanks for your cooperation and your helpful guide 
warm regards

FarshidAhmadi, 2017-08-14:
--------------------------------------------

 

 
 
 
 
 
 
 it on GitHub, or mute
 the thread.

TheSOMGuy, 2017-08-14:
You're welcome.

And btw. there is maybe a solution for this problem. 
If you track more than one person in a session you may could use a  ( or other feature-)Tracker for each of them. 
It's time consuming and needs a bit more ressources but it would help you.

Cheers

FarshidAhmadi, 2017-08-15:
would you please show me a simple example so how to use  time consuming in the frame ?
so i will work on it and i will consult with you about that 

many thanks for helpful guide

avpai, 2018-01-26:
I have a similar problem. I would like to track only the body that is the closest to kinect. I am using pykinect2 and I am also new to python. Any ideas or suggestions?