Kinect/PyKinect2 #30: Anyone else facing this issue?

karansaxena, 2016-06-17:
AttributeError: 'PyKinectRuntime' object has no attribute 'infrared_frame_desc'

gdauphin, 2016-07-18:
Same problem here :( . I also just tried a simple code to check "self._kinect.has_new_infrared_frame()" and always get false.
Works perfectly with depth and color but not with infrared.

And 'infrared_frame_desc' is in pykinectruntime.py ...

nseymoursmith, 2016-08-02:
I had this problem and fixed it by replacing the library files with the latest directly from this repo.

In my installation they were located at 
> anaconda3/lib/site-packages/pykinect2
Or somesuch similar place. You can find it by going into interactive mode and typing 
> import pykinect2
> help(pykinect2)

gdauphin, 2016-08-03:
It works perfectly ! Thank you for the tip :)

karansaxena, 2016-08-08:
@nseymoursmith This worked! Thanks!

closed 2016-08-08 by karansaxena

nmadapan, 2017-06-07:
Thank you. It works..

hayatkhan8660-maker, 2021-10-01:
> I had this problem and fixed it by replacing the library files with the latest directly from this repo.
>
> In my installation they were located at
>
>
> Or somesuch similar place. You can find it by going [...]

Thank buddy, it Worlks!