Kinect/PyKinect2 #36: ImportError: cannot import name PyKinectV2

DavidYaonanZhu, 2017-02-13:
I used Pip for install, and it worked.
but when I am running examples, the anaconda prompt show the error, ImportError: cannot import name PyKinectV2.
How can I solve?

aegorfk, 2018-08-03:
Hello! I have the same issue. Did you solve this?

aegorfk, 2018-08-03:
@DavidYaonanZhu Hello! I have the same issue. Did you solve this?

norlesh, 2018-12-20:
assuming this is the same import error I received caused by the "assert sizeof(tagSTATSTG) == 72" on line 2216 of PyKinectV2.py then the quick fix is to edit the file to "assert sizeof(tagSTATSTG) >= 72".

My assumption is that an extra field has been added to the structure in the Microsoft API... by convention Microsoft have usually/always append any new fields to the end of structures so as not to break backwards compatability... anyhow it allowed me to succefuly import PyKinectV2.

The more long term fix is for someone (who isn't me because I don't need any more distractions sorry) should find what the new field is and add it to the PyKinect2 API ... perhaps easier is just to modify all the asserts to >= to guard against future API changes upstream from Microsoft - not that I believe there will be any since KinectV2 is no longer something MS supports.