Kinect/PyKinect2 #37: AssertionError: 80

DavidYaonanZhu, 2017-02-13:
File "C:\Users\David\Anaconda2\lib\site-packages\pykinect2\PyKinectV2.py", line 2216, in <module>
    assert sizeof(tagSTATSTG) == 72, sizeof(tagSTATSTG)

nitzel, 2017-03-29:
You've probably installed via `pip` and are using x64 Anaconda or similar.

Go to the pykinect2 installation in the site-packages folder (`Lib\site-packages\pykinect2`) and replace the `.py` files with the ones from the github repository.

That worked for me, at least.

-----

It would be fantastic if the creator could update the pip package. @vladkol

hmnaik, 2017-06-27:
nitzel, thanks the solution works for me.

shrirangakadam007, 2017-09-15:
thanks it worked for me too

wanggilbert, 2018-04-17:
Thanks @nitzel

GARENFEATHER, 2018-04-22:
Thanks @nitzel

DQ0408, 2018-08-31:
@nitzel You f**king genius!!! THX~

autonomousvehicle123, 2019-05-03:
Thanks Dude!!

lemoncalamitous, 2020-01-16:
Works fine to me. Thanks!

Any thoughts on when to fix the pypi package version for ease of use in the future?

nitzel, 2020-01-16:
@lemoncalamitous Great. I'm pretty sure this repository is completely abandoned by it's owner @vladkol - I've asked for an updated pypi package nearly three years ago but to no avail - so don't expect it. 
I no longer have a Kinect around, otherwise I might've forked it and started maintaining it.

To work around this limitation, you can zip up the fixed python package and install it via `pip install ./pykinect2.zip` if you need it for an automated deployment.

arm235, 2020-04-14:
I am using python 2. I have done as per your resolution but still getting below error:
Traceback (most recent call last):
  File "ex-playback.py", line 44, in <module>
    import PyKinectTk
  File "C:\Python27\Lib\site-packages\PyKinectTk\__init__.py", line 61, in <module>
  File "C:\Python27\lib\site-packages\PyKinectTk\Capture\__init__.py", line 1, in <module>
    from DataCapture import *
  File "C:\Python27\lib\site-packages\PyKinectTk\Capture\DataCapture.py", line 11, in <module>
    from ..utils.PyKinect2 import *
  File "C:\Python27\lib\site-packages\PyKinectTk\utils\PyKinect2\__init__.py", line 1, in <module>
    from PyKinectV2 import *
  File "C:\Python27\lib\site-packages\PyKinectTk\utils\PyKinect2\PyKinectV2.py", line 2214, in <module>
    assert sizeof(tagSTATSTG) == 72, sizeof(tagSTATSTG)
AssertionError: 80

Please help to resolve this

KonstantinosAng, 2020-04-14:
maybe it's a python 2 problem, have you tried python 3?

KonstantinosAng, 2020-04-14:
I checked it with Python 2.7.16 and works, make sure that you have python 2.7 and higher. If it doesn't work uninstall and start from the beginning and make sure that you change the files in the libs/sites-packages/pykinect2.

arm235, 2020-04-17:
Yes. I am using 2.7.17 version of python. Tried with python 3 but I think code is using python2. Is there any other suggestion.

Still getting below issue:
(py2) C:\Users\741467\Downloads\PyKinectTk-master\Examples>python ex-playback.py -GUI
Traceback (most recent call last):
  File "ex-playback.py", line 44, in <module>
    import PyKinectTk
  File "C:\Users\741467\AppData\Local\Continuum\anaconda3\envs\py2\lib\site-packages\PyKinectTk\__init__.py", line 61, in <module>
    import Capture
  File "C:\Users\741467\AppData\Local\Continuum\anaconda3\envs\py2\lib\site-packages\PyKinectTk\Capture\__init__.py", line 1, in <module>
    from DataCapture import *
  File "C:\Users\741467\AppData\Local\Continuum\anaconda3\envs\py2\lib\site-packages\PyKinectTk\Capture\DataCapture.py", line 11, in <module>
    from ..utils.PyKinect2 import *
  File "C:\Users\741467\AppData\Local\Continuum\anaconda3\envs\py2\lib\site-packages\PyKinectTk\utils\PyKinect2\__init__.py", line 1, in <module>
    from PyKinectV2 import *
  File "C:\Users\741467\AppData\Local\Continuum\anaconda3\envs\py2\lib\site-packages\PyKinectTk\utils\PyKinect2\PyKinectV2.py", line 2214, in <module>
    assert sizeof(tagSTATSTG) == 72, sizeof(tagSTATSTG)
AssertionError: 80

(py2) C:\Users\741467\Downloads\PyKinectTk-master\Examples>python -version
Unknown option: -e
usage: python [option] ... [-c cmd | -m mod | file | -] [arg] ...
Try `python -h' for more information.

(py2) C:\Users\741467\Downloads\PyKinectTk-master\Examples>python --version
Python 2.7.17

KonstantinosAng, 2020-04-17:
I don't know any other solution. Maybe start for the start and create a new environment with Python 2.7.17 and replace again the PyKinectV2.py and PyKinectRuntime.py files. Easiest way to do this is to use anaconda to isolate and start with a fresh version of python.

arm235, 2020-04-17:
Have you tried with extracting features from xef files?
I am able to open GUI now but facing below issue:

Exception in Tkinter callback
Traceback (most recent call last):
  File "C:\Users\741467\.conda\envs\py2\lib\lib-tk\Tkinter.py", line 1547, in __call__
    return self.func(*args)
  File "C:\Users\741467\.conda\envs\py2\lib\site-packages\PyKinectTk\Playback\Select.py", line 148, in to_video
    pid = self.performance_id()
  File "C:\Users\741467\.conda\envs\py2\lib\site-packages\PyKinectTk\Playback\Select.py", line 99, in performance_id
    return self._performances[self.selected()]
  File "C:\Users\741467\.conda\envs\py2\lib\site-packages\PyKinectTk\Playback\Select.py", line 87, in selected
    return int(self._listbox.curselection()[0])

MichalPisanczuk, 2023-07-27:
Hi, After replaced files i have errors:
Traceback (most recent call last):
  File "d:/Pisak/objetosc-PY/Przykłady/openKinect.py", line 3, in <module>
    from pykinect2 import PyKinectV2
  File "C:\Users\it\AppData\Local\Programs\Python\Python36\lib\site-packages\pykinect2\PyKinectV2.py", line 2868, in <module>
    from comtypes import _check_version; _check_version('')
  File "C:\Users\it\AppData\Local\Programs\Python\Python36\lib\site-packages\comtypes\__init__.py", line 121, in _check_version
    raise ImportError("Wrong version")
ImportError: Wrong version

Python version: Python 3.6.4
Pip version: pip 21.3.1
Environment: Windows 11 64bit

Any solution?

MichalPisanczuk, 2023-07-27:
I solved problem
Installed comtypes==1.1.4

Misash, 2023-09-28:
> ty I solved problem Installed comtypes==1.1.4