Kinect/PyKinect2 #38: Get Z coordinate from body joint x and y

TheSOMGuy, 2017-07-11:
After playing around a while with the pyKinect BodyGame example and doing some research i didn't figure  out how to receive the z value by using the x and y coordinate from the body joints.

In [#31](https://github.com/Kinect/PyKinect2/issues/31) you wrote:
> You need to get z from depth frame using x and y you got from body_joints_to_depth_space 

what makes sense but this function doesn't exist anymore ( in the PyKinectRuntime.py ).

I've managed to receive the depth frame and to get the x and y coordinates like 
`joint_points[PyKinectV2.JointType_Head].x` 
and respectively  
`joint_points[PyKinectV2.JointType_Head].y`
but what's the meaning of this values? ( As an example x: 1529.32343242342 and y: 125.3425425 )
They don't fit as index for a standard image matrix ( i could crop them but i'm not shure if this is the "good way" ).

Can u explain how i reach my goal to receive the z value and would it be possible to write an API documentation as mentioned in other Issues before?
At least a list of functions with a small explanation how to use them would be enormous useful.

Thanks for the reply

TheSOMGuy, 2017-07-11:
The solution for the z-Question is quite simple: The pip repository is out of date. 
The question for a documentation still exists.

Thanks. Thread can be closed.

wxy656, 2017-08-08:
@TheSOMGuy 
z = self._depth[ y * 424 + x ]    the depth frame is 512*424
But I have another question:  is   the x,y value  color_frame's pixel or  a point in coordinate system? and is x,y value's unit   consistent with z value's (all meter？)

TheSOMGuy, 2017-08-08:
@wxy656 

It depends on the function call.
The call `depth_points = self._kinect.body_joints_to_depth_space(joints)` should always return values in the world coordinate system. 
But if i need the z coordinate of an image pixel i would use `joint_points = self._kinect.body_joints_to_color_space(joints)` for x and y.
Otherwise i obviously can't access the position in the image matrix.

Yes, they should be consistent.

dhecloud, 2018-09-06:
@TheSOMGuy  Hey, I been reading through your comments, and i figured i would have a better chance asking you since the repo seems to be abandoned.

I am interested in getting the depth images from a kinectv2.

What is the difference between `self._kinect.body_joints_to_depth_space` and `frame = self._kinect.get_last_depth_frame()`? Is there a need to use the former if the latter already returns the depth image? 
Also, do you have any tips on how to extract only the hand depth image from the full image? Right now i am setting depth values > and < than a threshold. it looks like this 
[image]. Not very sure if this is the best method

nitzel, 2018-09-06:
@dhecloud If you know a point that definitely belongs to the hand you can start region growing from there, adding nearby pixels that aren't too far away. That way you wouldn't have the two other artefacts that happen to be within your distance-range. You could also configure region growing to have a maximum depth-distance of say

mouarcher, 2018-11-29:
if self._bodies is not None:
                    if self._kinect.has_new_depth_frame: 
                        for i in range(0, self._kinect.max_body_count):
                            body = self._bodies.bodies[i]
                            if not body.is_tracked: 
                                continue 
                            
                            joints = body.joints 
                            # convert joint coordinates to color space 
                            joint_points = self._kinect.body_joints_to_color_space(joints)
                            self.draw_body(joints, joint_points, SKELETON_COLORS[i])
                            # get the skeleton joint x y z
                            depth_points = self._kinect.body_joints_to_depth_space(joints)
                            x = int(depth_points[PyKinectV2.JointType_ElbowLeft].x)
                            y = int(depth_points[PyKinectV2.JointType_ElbowLeft].y)
                            _depth = self._kinect.get_last_depth_frame()    
                            z = int(_depth[y* 512 + x] )
                            array_x.append(x)
                            array_y.append(y)
                            array_z.append(z)

i want to get the z coordinate, but there is an error (TypeError: 'NoneType' object is not subscriptable ), can anyone helps me~ thanks~

juanmaro97, 2019-11-28:
> 
> 
> @TheSOMGuy
> z = self._depth[ y * 512 + x ] the depth frame is 512*424
> But I have another question: is the x,y value color_frame's pixel or a point in coordinate system? and is x,y value's unit consistent with z value's (all meter？)

I do not understand the equation, may you explain me, please? Why 512?

nitzel, 2019-11-29:
The Kinect camera is taking three different kind of images. One of them is a colour image as you would get from your smartphone (with a resolution of 1920x1080 IIRC), an infrared image (can't remember the resolution) and a depth image with a resolution of 512x424 pixels.

The depth image is a single array of 512*424 numbers, one after another. The first pixel in that array is the pixel in the top left corner. The next pixel in the array is the pixel right next to it.
So if you wanted to iterate over the first row from the image, you'd do `for index in range(0, 512): self._depth[index]`.

Now if you would like to get a specific pixel where you know the row(=y) and column(=x) number of, then you can calculate its index via `index = row * 512 + column` since you need to sum up the number of pixels in all the rows above and then add the number of pixels to the left of the pixel you want.