Kinect/PyKinect2 #42: How to get the Point Cloud

Roios, 2017-09-11:
Hello,

How can I get the point cloud of the all image?The coordinates x,y and z of each pixel?
And the respective rgbd match?

Thanks

h-jia, 2018-01-22:
Up. Also curious about the point cloud

avpai, 2018-04-07:
Same here. Would really like to know how to get point cloud. If anyone got any ideas, please share.

sarabsethi, 2018-04-25:
me too, please!

LangeRobert, 2021-01-08:
Use the depth image as source of the polar points.

x = r * sin(a)*cos(b)
y= r * sin(a)*sin(b)
z= r*cos(a)

The angles a and b are results of the index and the FOV(70.6° x 60°):

```
for w in range(width):
    a = FOV_H * (i / width) - (FOV_H / 2)
    for h in range(height):
        b =FOV_V * (i / height) - (height / 2)
```