Kinect/PyKinect2 #44: PyKinectBodyGame just shows black screen

ChrisErgo, 2017-10-06:
I have a Kinect V2 that works with Processing, but I would like to use it with Python 3 Anaconda 64-bit. I realize that 64-bit is not very well tested, but has anyone gotten it to work?

TheSOMGuy, 2017-10-09:
Yepp.

I use the 64 bit version of Anaconda on my windows notebook as base for a virtual environment where the pykinect/pyGame example is running with Kinect V2.
Can you tell me what your problem is?

ChrisErgo, 2017-10-09:
I started with trying to get PyKinectBodyGame to work. It opens a large black window with the title "Kinect for Windows v2 Body Game", and that's all it does.

"pip install pykinect2" didn't get the latest versions of PyKinectRuntime.py and PyKinectV2.py, leading to a failed assertion for "assert sizeof(tagSTATSTG)" so I copied those files in manually to get past that.

TheSOMGuy, 2017-10-10:
Did you follow the installation instructions on the pyKinect2 [github](https://github.com/Kinect/PyKinect2) site?

ChrisErgo, 2017-10-11:
The game works now. Thank you for the tips. You made me realize that I had previously installed libusbK drivers (using zadig) to get the Kinect to work with Processing. Uninstalling those and letting the SDK drivers install themselves fixed the problem. That should get me rolling with my own development. Thank you.

closed 2017-10-11 by ChrisErgo