Kinect/PyKinect2 #46: How to capture images ?

dhruvdesmond, 2018-01-10:
I am doing a project to track human gait. How to use this library to get images from the feed to create a data set ?

TheSOMGuy, 2018-01-17:
This is a strange question, but anyway.

First of all you have to run python and stuff to get this working. ( See the Readme section for installation instructions. )
A very good idea is to create and use a virtual environment for your PyKinect stuff.
( [see the virtualenv section on this page](http://docs.python-guide.org/en/latest/dev/virtualenvs/) )
After that you should us the PyKinectBodyGame.py in the examples folder to test if PyKinect2 is running.
If it works you have to implement the routines for your dataset generation by yourself.
I would tell you more, but i don't know which data u need ( depth, color etc. ).
So if you need more information write more details about the dataset you want to create.

Have fun.

sangk54, 2019-05-11:
Hi,
I need to get depth information or 3D point cloud. Could you send me some example code ? Thanks !

nitzel, 2019-05-16:
@sangk54 take a look at my code in issue #72 where I have a class that returns an array of camera space points (3d coordinates).