Kinect/PyKinect2 #47: how to save skeleton data?

h-jia, 2018-01-23:


TheSOMGuy, 2018-01-25:
Your initial question is "how to save skeleton data" and then you ask how to print them.
Can you please specify what exactly you want to do?

Thanks.

sxs4337, 2018-02-09:
I have a similar issue so let me try to elaborate based on the original question.

I am trying to dump to disk the following- the raw RGB frame as image, infrared frame as image and the skeletal points as text file.
Any related examples would be vert helpful.
Thanks in advance.

TheSOMGuy, 2018-02-10:
To store the raw RGB frame u can use PIL and the im.save function.
The matrix A have to be an numpy array.
And this should also work for the infrared frame if u store the infrared value in a RGB-Tripel ( infaredValue, infaredValue, infaredValue ).
Then u can also build a numpy array A from the infrared RGB-Tripel matrix and store it with PIL.
So, if A holds your data ( rgb or infrared )

```
import numpy
from PIL import Image

RGB = []

if( A == infrared )
for item in A: RGB.append([item,item,item])
RGB = numpy.array(RGB)
else:
RGB = numpy.array(A)

im = Image.fromarray(RGB)
im.save("your_file.jpeg")

```
This should work for the infrared and RGB data.
The algorithm for the skeleton depends on what exactly do you want to store.

sxs4337, 2018-02-10:
Thanks a lot for the reply!
I guess the trouble is on how to get the frame values in variable A. (as in the the step before your code above)

I am reading through the examples and at line-
https://github.com/Kinect/PyKinect2/blob/master/examples/PyKinectInfraRed.py#L83

I get the following values-
(Pdb) p frame
array([   0, 2247, 1649, ..., 8948,  794,    0], dtype=uint16)
(Pdb) p frame.shape
(217088L,)

The frame dimensions are 512*424 = 217088. Can we simply resize this to get the values and then follow from your instructions above to dump it as an image.

TheSOMGuy, 2018-02-12:
Not really.

You have to compute a RGB like representation if you want to store them as an .jpg or .png or something.
This means you have to map the infrared value for each (x,y) image grid position to a range between 0 and 1 ( RGB double triple ) or to a range between 0 and 255 ( RGB int tripel ). 
Do something like this:

```
min = math.min(frame)
max = math.max(frame)

# For double RGB triple
mmRange =  1 / ( max - min )

# For int RGB triple
mmRange =  255 / ( max - min )

# Classic pythonic way of vector multiplication (  list comprehension ) 
frame = [ x * mmRange for x in frame ]
# or using numpy
frame = numpy.multiply( numpy.array(frame), mmRange )

```
Then store the frame as .png or .jpg or something else using the methods described above.

sxs4337, 2018-02-12:
That worked! Thanks.
I was able to dump color and IR as PNG files and the corresponding body joints data into a CSV file.

juanmaro97, 2019-12-02:
> 
> 
> That worked! Thanks.
> I was able to dump color and IR as PNG files and the corresponding body joints data into a CSV file.
> 
> Here is my version-
> https://github.com/sxs4337/PyKinect2/blob/master/examples/PyKinectSave.py

You didn't save the Z coordinate data here, did you?