Kinect/PyKinect2 #48: Track the closest body

avpai, 2018-01-23:
Hi,
I am new to python and kinect. I have been studying the example given here,as there are no other tutorials/documentation. I am able to get the depth of a joint (any joint) of the bodies tracked by the kinect. But I would like to track only the closest body(based on the depth calculated). Any ideas or suggestions? Thank you so much in advance.

TheSOMGuy, 2018-01-25:
As a suggestion, compute the center of mass of a skeleton. 
Sum up all joints in 3D and compute a mean position.  Then compare the centers. 
The closest center to the camera plane is the skeleton you looking for. 

But a question from my side, how did you get the depth of the joints?
Thanks for reply.

avpai, 2018-01-25:
@TheSOMGuy - Thank you so much for your reply.

                 joint_points_depth = self._kinect.body_joints_to_depth_space(joints)
                    x= int(joint_points_depth[PyKinectV2.JointType_Head].x)
                    y= int(joint_points_depth[PyKinectV2.JointType_Head].y)
                    z = int(self._depth[ y * 512 + x ] ) ## Its in Millimeters
                    z = int(z/(25.4)) ##Its in Inches 

I got the depth using the above method.  Could you provide a sample snippet on how to compare the depths of all bodies and the choose just one?  I tried the below code in the same for loop:

                     track[i]= self._body_depth
                    closest= min(track.items(),key = lambda x:x[1])
                    print(closest)
but I am not getting the right answer.I was planning to get the tracking id or something and then use it to track that body which is the closest.

avpai, 2018-02-08:
@vladkol , @benlower , guys, what do think?

avpai, 2018-02-23:
@TheSOMGuy. I go the individual distances and this is the code:

> for i in range(0, self._kinect.max_body_count):                    
                   
                    body = self._bodies.bodies[i]                    
                    if not body.is_tracked: 
                        continue 
                   
                    joints = body.joints            
                    z = self.distance_body(joints) # gets the distance
                    track[i]=z
                    print(track)
                    min_dist= min(track.values())
                    for k in track.keys(): 
                        if track[k] == min_dist:
                            print(k)
                            l=k
                            
                    if(i==l):                                          
                        body_near = self._bodies.bodies[l]              
                        print(i)                      
                        joints_near = body_near.joints
                        joint_points_near = self._kinect.body_joints_to_color_space(joints_near)
                        joint_points_depth_near = self._kinect.body_joints_to_depth_space(joints_near)
                        self.draw_body(joints_near, joint_points_near, SKELETON_COLORS[l])
                     else:
                        continue

closed 2018-02-28 by avpai