Kinect/PyKinect2 #56: Mapper.MapDepthFrameToColorSpace 'The parameter is incorrect'

muharremmandaci, 2018-07-02:
depthframe = kinect.get_last_depth_frame()
ptr_depth = numpy.ctypeslib.as_ctypes(depthframe.flatten())
L = kinect._depth_frame_data_capacity.value

S = 1080 * 1920
TYPE_ColorSpacePointArray = PyKinectV2._ColorSpacePoint * S
csps1 = TYPE_ColorSpacePointArray()

x = kinect._mapper.MapDepthFrameToColorSpace(L, ptr_depth, S, deneme)

i have an error => 'The parameter is incorrect'

danvica, 2018-08-24:
Hi muharremmandaci, did you solve the issue ?

Lancename, 2019-05-02:
kinect = PyKinectRuntime.PyKinectRuntime(PyKinectV2.FrameSourceTypes_Color | PyKinectV2.FrameSourceTypes_Depth | PyKinectV2.FrameSourceTypes_Infrared)        
color2depth_points_type = _DepthSpacePoint* np.int(1920*1080)
color2depth_points = ctypes.cast(color2depth_points_type(), ctypes.POINTER(_DepthSpacePoint))
kinect._mapper.MapColorFrameToDepthSpace(ctypes.c_uint(512 * 424), kinect._depth_frame_data, ctypes.c_uint(1920 * 1080), color2depth_points)  
 i used this code and it is fine ,then what we want is the 'color2depth_points'. if you want to take points from color space to depth_space (such as [255,255] in color space),just use color2depth_points[255*1920+255-1].x   and  color2depth_points[255*1920+255-1].y  to take the points in depth space.

StaringtheStar, 2019-08-01:
@Lancename , first thx for your code. It works without any error!
Though I have some problems right now.
I got a data from your code, but after I don't know how to use that. Can you give me a small help if you don't mind..?