Kinect/PyKinect2 #57: unknown encoding

aegorfk, 2018-08-03:
Whole running import lines right after installation:
```
from pykinect2 import PyKinectV2
from pykinect2.PyKinectV2 import *
from pykinect2 import PyKinectRuntime
```

I got:
  File "<string>", line unknown
SyntaxError: unknown encoding for '....../lib/python3.5/site-packages/pykinect2/PyKinectV2.py': mbcs

I tried even manually rewrite the file, but did not help. Any ideas?

aegorfk, 2018-08-05:
Actually, it happened because this package is so Windows-based systems, not Unix. 
I understood.
Solved

closed 2018-08-05 by aegorfk

navinthenapster, 2018-10-23:
how can I make it available for Linux based systems. Any idea

aegorfk, 2018-10-24:
@navinthenapster I used pylibfreenect2 package for Mac integration.
You can see it here: https://github.com/aegorfk/Mask_RCNN/blob/master/Kinect_streaming_segmentation.ipynb