Kinect/PyKinect2 #59: How to segment the human body in front of the kinect camera using PyKinect2?

soveida24, 2018-09-18:
I use PyKinect2 for connecting Kinect camera. I run the PyKinectInfraRed example and change it for depth. Now I want to segment just body. I think it's better to use distance . and if the distance is more than a certain value, they are removed. Now I write like this but it doesn't work correctly.
could you help me?
Thank you

 def draw_depth_frame(self, frame, target_surface):
        target_surface.lock()
        f8 = np.uint8(frame.clip(1, 4000) / 16.)
        frame8bit = np.dstack((f8, f8, f8))
        address = self._kinect.surface_as_array(target_surface.get_buffer())
        print(frame8bit.ctypes.data)
        if(frame8bit.ctypes.data<1053492053536):
            ctypes.memmove(address, frame8bit.ctypes.data, frame8bit.size)
        del address
        target_surface.unlock()

avpai, 2019-03-19:
Hi,

I am stuck at the same position, did u manage to solve it?

Sourab1801, 2022-05-18:
hi @avpai and @soveida24 
 I am also stuck here, if you found any solution?

Sourab1801, 2022-05-18:
hi @soveida24  & @avpai 
  I am also stuck here, you found any solution regarding this?