Kinect/PyKinect2 #60: X,Y Joint coordinates- OverflowError: cannot convert float infinity to integer

avpai, 2018-09-22:
When I come close to the sensor , I get an error saying
" OverflowError: cannot convert float infinity to integer"

for the lines of code which I use to get the x,y,z co-ordinates of the joints. I tried this:
  ```
def round_int(self,x):         
        if x == float("inf") or x == float("-inf"):
            return 0 # or x or return whatever makes sense
        else:
            return int(round(x))
```
and used this function on the below sample or stuffs like that that gives me a x,y coordinate
`rightelbow_y = self.round_int(int(joint_points[PyKinectV2.JointType_ElbowRight].y))`

But I am still getting this error. Any ideas or suggestions to improve this, will be much appreciated

nitzel, 2018-09-22:
General idea: surround with try catch and in the catch take a closer look at the value in question to see why it doesn't equal float("inf")

nitzel, 2018-09-22:
Your problem probably lies in the int(...) cast which I removed here:

rightelbow_y = self.round_int(joint_points[PyKinectV2.JointType_ElbowRight].y)