Kinect/PyKinect2 #68: Opencv and Masking

avpai, 2019-03-18:
Hi Guys,

So, I saw a lot of posts and examples about using OpenCV and PyKinect2 library. I have some questions and I hope someone will help me solve them.
[image]

So I got a binary mask and I want to binary_and it with the tracking frame. Any ideas on how to do it?
I saw the examples bu the frame that I got from it was the regular color frame not the tracking drawn one.

yxp1003, 2019-03-19:
Sorry, I don't quite understand what you mean. Do you mean that you only want bone tracking images and not color frames? Maybe u can convert the frame toString and then processing.

avpai, 2019-03-20:
The RGB image is from the Pykinect2 example. I was hoping to  use OpenCV on it, like you did. By saving it as a .png file and using opencv to read it.  But it is not smooth. Did you mange to find a work around for this problem?

yxp1003, 2019-03-21:
> RGB图像来自Pykinect2示例。我希望像你一样使用OpenCV。将其保存为.png文件并使用opencv进行读取。但它并不顺利。你有没有找到解决这个问题的方法？

I have tried save .png and loaded it, but it is not a good a idea, becuz the frame looked not  fluency.
    def pygame_to_cvimage(self, surface_to_draw):
        image_string=pygame.image.tostring(surface_to_draw, 'RGB')
        image_np = np.fromstring(image_string, np.uint8).reshape(405, 720, 3)
        img = cv2.cvtColor(image_np, cv2.COLOR_BGR2RGB)
        return img
this fucntion can vonvert the frame to a String type, and OpenCV can use it