Kinect/PyKinect2 #72: Pykinect2 + OpenGL?

avpai, 2019-05-14:
Has anyone tried integrating OpenGL to Pykinect2 to get a 3D view? Is it possible since Pykinect2 doesn't provide point cloud data?

nitzel, 2019-05-15:
It should be possible. I've plotted my point clouds that I got from PyKinect2 via matplotlib (which has some basic support for 3d plots).

PyKinect can indeed give you 3D-coordinates, so I would call that a point cloud.

avpai, 2019-05-16:
So  use something like:

`fig = plt.figure()
 ax = plt.axes(projection='3d')
  X,Y,Z = self.map_depth_frame_to_camera_space_points                        
  ax.plot_surface(X, Y, Z, rstride=1, cstride=1, cmap='viridis', edgecolor='none');`

nitzel, 2019-05-16:
`depth_to_csp_buffer` is not part of the MS Kinect class, I've added that to my class manually. (see the `__init__` constructor).

You can define `TYPE_CameraSpacePoint_Array = PyKinectV2._CameraSpacePoint * (self.WIDTH * self.HEIGHT)` and use `TYPE_CameraSpacePoint_Array` instead of `self._kinect.depth_to_csp_buffer`.

avpai, 2019-05-16:
hi,
I removed the errors, but I could'nt get it to plot. Is a sample code or code snippet that you could share for the plotting array?

nitzel, 2019-05-17:
Have you tried plotting the 2D projection? So only the X/Y coordinates of each pixel. Then you should already be able to recognise what's in front of your camera. If that does work, move on the the 3D plotting which is a whole lot tricker as you have to get the viewport, angle, etc right.

```python
...
csps = my_kinect.map_depth_frame_to_camera_space_points()
```
These csps are basically `x1,y1,z1,x2,y2,z2,...` not `[[x1, x2,...][y1, y2, ...][z1, z2, ...]]`.
So `X,Y,Z = csps` doesn't work.

I'm sorry, I haven't touched this code since 2017 and no longer have a Kinect to test code against - so no examples and no recently tested code from me.

KonstantinosAng, 2020-04-27:
I just created a repository that uses the PyQtGraph, OpenGL and PyKinect2 to create real - time point clouds like the KinectSDK. It is fully optimized, using numpy that runs in C and can create point clouds up to 60+ frames. It can create point clouds using the depth camera, the rgb camera, only the body index frame or the skeleton track joints. Also it can simultaneously create all of the above point clouds.
In addition, there are 4 track bars that can be used to change the point size and color.

Check it out, and if you like it maybe give it a star to keep track of the updates.

https://github.com/KonstantinosAng/PyKinect2-PyQtGraph-PointClouds

If you cannot understand how to use it or if you run into any problems feel free to ask me.

lixryjz, 2024-06-15:
when I use MapDepthFrameToCameraSpace，I find it get points all is -inf？How to salve it？

nitzel, 2024-06-15:
@lixryjz doe the values you get look same before yoiu do `MapDepthFrameToCameraSpace`? If so, check you're calling `MapDepthFrameToCameraSpace` with the right arguments. If not, maybe your Kinect is broken.

However, I haven't touched this project since 2017, so I'll be unsubscribing from this thread as I don't remember enough details to help ^^