Kinect/PyKinect2 #75: Multiple kinects

kirivasile, 2019-10-23:
Hi everyone! Is it possible to use depth images of two connected kinects?
I've managed to make them work in http://docs.ipisoft.com/Multiple_Kinects_v2_on_a_Single_PC
But don't know how to do it in this project. 

I can only view the first frame of connected kinect.

nitzel, 2019-10-23:
Hi,
I would assume that you'll have to change the runtime so that it can be configured to choose a different connect than the default. Probably affects this line:
https://github.com/Kinect/PyKinect2/blob/c86d575175edf5fc0834590acdfb82b55b5ccd96/pykinect2/PyKinectRuntime.py#L65

Have a look at the C# Documentation of the Kinect2 library to see how it could be done.

kirivasile, 2020-02-17:
Haven't found any solution yet. I believe it's not supported.
It's better to use multiple computers for this purpose.

closed 2020-02-17 by kirivasile

KonstantinosAng, 2020-02-17:
i believe that [pylibfreenect2](https://github.com/r9y9/pylibfreenect2) (runs only on Linux and not in Windows) can access multiple kinect's  if the hardware supports it, meaning that you have more than one usb 3.0 port.