Kinect/PyKinect2 #77: How to get the 3d coordinate of  the Joint_Points?

JohnleeHIT, 2019-12-31:
Any one know how to get the 3d coordinate of  the Joint_Points? we can get the 2D coordinate of the Joint_points by using :
           joint_points = self._kinect.body_joints_to_color_space(joints)
Generally, we can get the 3d coordinate from the object "joints" , However,  I just can't get the attribute from the object when debugging.

KonstantinosAng, 2020-01-29:
if you find a body and write:

joints = body.joints

Then to get the world coordinates of any joint:
Example for Right Hand
x = joints[PyKinectV2.JointType_HandRight].Position.x
y = joints[PyKinectV2.JointType_HandRight].Position.y
z = joints[PyKinectV2.JointType_HandRight].Position.z

all points are in meters

JohnleeHIT, 2020-01-30:
> if you find a body and write:
>
> joints = body.joints
>
> Then to get the world coordinates of any joint:
> Example for Right Hand
> x = joints[PyKinectV2.JointType_HandRight].Position.x
> y = [...]

Got it ,thanks!

jmatthess, 2021-06-05:
> if you find a body and write:
>
> joints = body.joints
>
> Then to get the world coordinates of any joint:
> Example for Right Hand
> x = joints[PyKinectV2.JointType_HandRight].Position.x
> y = [...]

This code here is almost perfect for my use case but im finding that the y position is giving me values that are negative. i imagine this is because the joint im tracking goes below the cameras current height. but i was wondering if its possible to get the "calibrated" values?

KonstantinosAng, 2021-06-07:
The calibration only works for undistorting the RGB or Depth Image and it will help you get accurate data when you perform your own image processing on the RGB or Depth Image. However, the Kinect's tracking algorithm that gives you the joint values cannot be calibrated as we don't know what Microsoft is doing behind it. I am pretty sure they have calibrated their data.

ahmedetihad, 2022-02-06:
> if you find a body and write:
>
> joints = body.joints
>
> Then to get the world coordinates of any joint: Example for Right Hand x = joints[PyKinectV2.JointType_HandRight].Position.x y = [...]

thanx that's help me