Kinect/PyKinect2 #78: How PyKinect2 can be used in Ubuntu. When Kinect WIndow SDK is for Windows

Archeema16, 2020-01-10:
Hello. I know this sounds immature but i am new to Kinect Community so i have no other way. 
I am currently using pylibfreenect2 to run kinect for my project but somehow i am unable to get depth Data. When started searching i stumbled upon an opensource code which uses PyKinect2 to get it. 
Now the problem is PyKinect2 get installed by pip in ubuntu but it has a prereq which is Kinect for Windows SDK V2 which i dont know how to install on ubuntu. As the material available for Kinect V2 is very outdated. It would be great if you help me on it.

nitzel, 2020-01-13:
I guess your best bet is having a look at OpenKinect. That's unlikely to be compatible with PyKinect2, though.

KonstantinosAng, 2020-02-04:
Maybe this one:

https://github.com/kiddos/pykinect2