Kinect/PyKinect2 #79: Include Additional Example Programs

Ddylen, 2020-01-29:
This library can be somewhat difficult to understand how to use (its not immediately clear how to use the functions in it with only the C++ SDK as a guide (https://docs.microsoft.com/en-us/previous-versions/windows/kinect/dn799271(v=ieb.10)), so additional example programs would be useful for familiarising programmers with the workings of the library. I've included below basic demonstration programs on how to save kinect data, and get 3D positions from pixel values, in the hope that other new starters might find it useful. Would it be possible to incorporate them into the examples folder?

In addition, given this project does not seem to be actively developed, if anyone else has example programs they think new starters might find useful, commenting them here could help make this thread itself a useful resource for people trying to figure out how to use the library. 

```
import numpy as np
import cv2
import pickle
import time 
import datetime

from pykinect2 import PyKinectV2
from pykinect2.PyKinectV2 import *
from pykinect2 import PyKinectRuntime

def save_frames(FILE_NAME):
    #records and saves colour and depth frames from the Kinect
    
    print("Saving colour and depth frames")
    
    # define file names
    depthfilename = "DEPTH." + FILE_NAME +".pickle"
    colourfilename = "COLOUR." + FILE_NAME +".pickle"
    depthfile = open(depthfilename, 'wb')
    colourfile = open(colourfilename, 'wb')
    
    #initialise kinect recording, and some time variables for tracking the framerate of the recordings
    kinect = PyKinectRuntime.PyKinectRuntime(PyKinectV2.FrameSourceTypes_Color | PyKinectV2.FrameSourceTypes_Depth)
    starttime = time.time()
    oldtime = 0
    i = 0
    fpsmax = 0
    fpsmin = 100
    
    display_type = "COLOUR"
    #display_type = "DEPTH"
    
    # Actual recording loop, exit by pressing escape to close the pop-up window
    while True:
        
        if kinect.has_new_depth_frame() and kinect.has_new_color_frame() :
            elapsedtime = time.time()- starttime
            if(elapsedtime> i/10):
                
                #Only for high i try evalutaing FPS or else you get some divide by 0 errors
                if i >10:
                    try:
                        fps =  1/(elapsedtime - oldtime)
                        print(fps)
                        if fps> fpsmax:
                            fpsmax= fps
                        if fps < fpsmin:
                            fpsmin = fps
    
                    except ZeroDivisionError:
                        print("Divide by zero error")
                        pass
                    
                oldtime = elapsedtime
                
                #read kinect colour and depth data (somehow the two formats below differ, think one is and one isnt ctypes)
                depthframe = kinect.get_last_depth_frame() #data for display
                depthframeD = kinect._depth_frame_data
                colourframe = kinect.get_last_color_frame()
                colourframeD = kinect._color_frame_data
                
                #convert depth frame from ctypes to an array so that I can save it
                depthframesaveformat = np.copy(np.ctypeslib.as_array(depthframeD, shape=(kinect._depth_frame_data_capacity.value,))) # TODO FIgure out how to solve intermittent up to 3cm differences
                pickle.dump(depthframesaveformat, depthfile)
                
[... 189 lines cut ...]
    out.write(frame)
    cv2.imshow('Stiching Video',frame)

    #Cycle through the rest of the colour frames, stiching them together
    while True:
        try:
            frame = pickle.load(datafile)
            out.write(frame)
            cv2.imshow('Stiching Video',frame)
            if (cv2.waitKey(1) & 0xFF) == ord('q'): # Hit `q` to exit
                break
        except EOFError:
            print("Video Stiching Finished")
            break

    # Release everything if job is finished
    out.release()
    cv2.destroyAllWindows()
    
if __name__ == '__main__': 
    
    #replace name below with the corresponding section of the name of your saved depth data (for reference, the full name of my saved colour data file was COLOUR.test.1.29.13.17.pickle)
    frames_to_video('test.29.1.13.23')

```

NIravMeghani, 2020-04-08:
how to plot this 3d points in any scatter plot ?

Ddylen, 2020-04-08:
> how to plot this 3d points in any scatter plot ?

Im not sure I understand the question- pixel_positions_3D_list is a list of lists of 3D points. The number of 'sub list's corresponds to the number of colour image pixels specified in colour_image_pixels_to_locate_list. Each individual sublist gives the 3D poistion of said pixel in camera coordinates in each recorded frame. 

As for plotting the 3D points in a sublist, you could do that with any common python plotting tool (e.g. https://matplotlib.org/3.1.1/gallery/mplot3d/scatter3d.html)

NIravMeghani, 2020-04-08:
https://drive.google.com/open?id=1x3xQvYzTp1XmJKhPq-2X2zmJ6dRhRXRL
if you find any error in this then tell me 
or any other way of you found then tell me.

Ddylen, 2020-04-08:
I'm somewhat timepressured right now so ive only skimmed through it, but what jumps out at me is that your seem to have defined your own function to find the 3D position from the 2D colour and 1D depth data. The Kinect library has a function for that directly, which I've called in the code above (see             x_3D = csps1[y*1920 + x].x
                y_3D = csps1[y*1920 + x].y
                z_3D = csps1[y*1920 + x].z) 
It requires a bunch of other things to be defined first, unfortunately i wrote this a while back so I cant recall which of the random things I define at the start of my program were required for this specific function

KonstantinosAng, 2020-04-16:
Check this issue for Creating 3d scatter plots:

https://github.com/Kinect/PyKinect2/issues/82