Kinect/PyKinect2 #81: How to get the depth camera intrinsics

ShuangjunLiu, 2020-02-15:
From the source code, I see there is method in PyKinectV2 to get depth camera intrinsics.  

However, I have no idea how to call this function.  
Can you give an example how to use these com methods?  

Thanks.

KonstantinosAng, 2020-02-15:
I have a repository with all the mapper functions if you want to check it out:

https://github.com/KonstantinosAng/PyKinect2-Mapper-Functions

For the depth camera intrinsics you can use the code below:

from pykinect2.PyKinectV2 import *
from pykinect2 import PyKinectV2
from pykinect2 import PyKinectRuntime

kinect = PyKinectRuntime.PyKinectRuntime(PyKinectV2.FrameSourceTypes_Depth | PyKinectV2.FrameSourceTypes_Color)

while True:
    if kinect.has_new_depth_frame():
        intrinsics_matrix = kinect._mapper.GetDepthCameraIntrinsics()

KonstantinosAng, 2020-02-15:
and to access the values use:

focal_length_x = intrinsics_matrix.FocalLengthX
focal_length_y = intrinsics_matrix.FocalLengthY
principal_point_x = intrinsics_matrix.PrincipalPointX
principal_point_y = intrinsics_matrix.PrincipalPointY
radial_distortion_fourth_order = intrinsics_matrix.RadialDistortionFourthOrder
radial_distortion_second_order = intrinsics_matrix.RadialDistortionSecondOrder
radial_distortion_sixth_order = intrinsics_matrix.RadialDistortionSixthOrder

ShuangjunLiu, 2020-02-15:
Thank you so much!
It works!

theonlyhacker, 2023-06-28:
How can I get the internal parameters of the kinect color camera? There should be a corresponding function, but I can't find it？I guess the color camera intrinsics function like GetColorCameraIntrinsics()，but it does not exist.

KonstantinosAng, 2023-06-28:
Yes there is only a function to get the Depth intrinsic parameters:

intrinsics_matrix = kinect._mapper.GetDepthCameraIntrinsics()

But you can always calibrate the RGB sensor and calculate the intrinsics yourself.