Kinect/PyKinect2 #85: Outputting no frames or data to screen. [URGENT]

RoboFinley, 2020-11-19:
I am trying to run the examples given. When I run PyKinectBodyGame I get no errors and the lights on the Kinect (Xbox One version) turn on but it doesn't detect any frames or output anything to the pygame window. I also get an error while running the PyKinectInfraRed example, an error stating AttributeError: 'PyKinectRuntime' object has no attribute 'infrared_frame_desc'.

I did have to make a change to line 2216 in the PyKinectV2, changing thesizeof(tagSTATSTG) == 72 to a <=, I also changed line 154 in PyKinectRuntime, changing the .clock() to a .process_time() because clock() is no longer supported as it was returning an error. 
The Kinect does work as it does show output using Kinect Studio. Am I missing something?
I am using the pycharm IDE and have downloaded pykinect2 through the python interpreter settings section, I have also tried downloading directly from github using the easy_install and the version still seems to be 0.1.0. This not only doesnt fix anything, it also means I cant make changes to the PyKinectRuntime file to solve the clock() error anymore as it is saying it has failed to make it writable.  

If I have not given some information needed to answer my question please let me know and I will add it in.

Thank you in advance for your help.

ukasu, 2021-01-07:
I have same problem. There no frames or data in pygame window.

WangXingfang, 2021-01-07:
I have the same problem. Do you have a solution?

ukasu, 2021-01-07:
I checked in has_new_color_frame function in PyKinectRuntime.py
When I run the example I seen that self._last_color_frame_time and self._last_color_frame_access value are the same. 
In IF-ELSE block it has to be self._last_color_frame_time bigger than self._last_color_frame_access. Otherwise the block return false and there is no data or screen in program

ukasu, 2021-01-07:
I solved the problem. The solution is that replace all time.clock() to time.perf_counter() in PyKinectRuntime.p
[image]
y

WangXingfang, 2021-01-08:
> I solved the problem. The solution is that replace all time.clock() to time.perf_counter() in PyKinectRuntime.py
> 
> [image]

Got it! Thank you very much.

mamoonik, 2022-08-08:
need help with this issue!!