Kinect/PyKinect2 #86: How to get the real-time 3D coordinates of a point,such as the end of a manipulator

WangXingfang, 2021-01-08:
Hello all,
Is there a way to get the real-time 3D coordinates of a point, such as the end point of a manipulator?
It would be great if you help me on it.

LangeRobert, 2021-01-09:
I am working on a solution wich uses polar to cartesian coordinate conversion using cython. The angels can be calculated threw the index. If i'm done i'll send the code in here.
 :)

caohena, 2021-01-13:
我也在用pykinect库去做些应用，方便留个联系方式么

WangXingfang, 2021-01-13:
> 我也在用pykinect库去做些应用，方便留个联系方式么

xingfangw@126.com

KonstantinosAng, 2021-01-23:
I have made real time point clouds with the Kinect. If you know the place of the manipulator you can isolate the pointcloud, otherwise you can put a marker on the robotic manipulator and track the location of the marker

https://github.com/KonstantinosAng/PyKinect2-PyQtGraph-PointClouds

closed 2022-04-15 by WangXingfang