Kinect/PyKinect2 #88: Importing PyKinectV2 throws ImportError("Wrong version")

vitrack, 2021-03-18:
When I try to import PyKinectV2 from pykinect2, it comes back with this:
```
Traceback (most recent call last):
  File "[Censored bc it includes my name]\PyCharm\Projects\PyKinectTest\test.py", line 1, in <module>
    from pykinect2 import PyKinectV2
  File "[Censored bc it includes my name]\PyCharm\Projects\PyKinectTest\venv\lib\site-packages\pykinect2\PyKinectV2.py", line 2868, in <module>
    from comtypes import _check_version; _check_version('')
  File "[Censored bc it includes my name]\PyCharm\Projects\PyKinectTest\venv\lib\site-packages\comtypes\__init__.py", line 29, in _check_version
    raise ImportError("Wrong version")
ImportError: Wrong version
```

KonstantinosAng, 2021-03-18:
Try this solution to see if it fixes it:

https://github.com/Kinect/PyKinect2/issues/37

vitrack, 2021-03-18:
@KonstantinosAng It does not.

KonstantinosAng, 2021-03-18:
You can always go to venv\lib\site-packages\comtypes\__init__.py in line 29 and delete the if statement to see if it works.

SujithChristopher, 2021-04-03:
@vitrack 
I had the same issue, 

it appears to be an issue with the comtypes version

try:
pip install comtypes==1.1.4

Hope it helps

marcocrxu, 2021-05-30:
i have also  met the problem. after try #37 the problem still exist. but i find that reboot solve my probelm.

wjrzm, 2022-04-15:
> @vitrack I had the same issue,
> 
> it appears to be an issue with the comtypes version
> 
> try: pip install comtypes==1.1.4
> 
> Hope it helps

It works!