Kinect/PyKinect2 #89: How to get real time x,y,z coordinates of a detected object using opencv

5av10, 2021-05-08:
How do I extract the real time X,Y,Z coordinates of a detected blob?
I am using opencv  with cv2.SimpleBlobDetector_create() to detect and track the ball in the dept video but cannot manage to sample a point from the detected blob and get its x,y,z coordinates the code is:

        depth_colormap= cv2.applyColorMap(cv2.convertScaleAbs(depth_img, alpha=255/clipping_distance), cv2.COLORMAP_JET)
        g=cv2.cvtColor(dept_colormap,cv2.COLOR_BGR2GRAY)
        detector=cv2.SimpleBlobDetector_create(params)
        points=detector.detect(g)
        blank=np.zeros((1,1,1))
        blobs=cv2.drawKeypoints(g,points,np.array([]),(0,255,0),cv2.DRAW_MATCHES_FLAGS_DRAW_RICH_KEYPOINTS)

KonstantinosAng, 2021-05-10:
Using my [mapper repo](https://github.com/KonstantinosAng/PyKinect2-Mapper-Functions) you can use the following code to get the world points of the detected blobs:

```

import mapper

depth_colormap = cv2.applyColorMap(cv2.convertScaleAbs(depth_img, alpha=255/clipping_distance), cv2.COLORMAP_JET)
g = cv2.cvtColor(dept_colormap,cv2.COLOR_BGR2GRAY)
detector = cv2.SimpleBlobDetector_create(params)
points = detector.detect(g)

for point in points:
  """ Get depth x, y points of detected blobs """
  depth_x, depth_y = point.pt[0], point.pt[1]
  """ Get world points of your depth x, y """
  world_x, world_y, world_z = mapper.depth_point_2_world_point(kinect, _DepthSpacePoint, [depth_x, depth_y])

blank=np.zeros((1,1,1))
blobs=cv2.drawKeypoints(g,points,np.array([]),(0,255,0),cv2.DRAW_MATCHES_FLAGS_DRAW_RICH_KEYPOINTS)

```

5av10, 2021-05-11:
> Using my [mapper repo](https://github.com/KonstantinosAng/PyKinect2-Mapper-Functions) you can use the following code to get the world points of the detected blobs:
>
> ```
>
> import mapper [...]

The z coordinate is not changing