Kinect/PyKinect2 #90: distortion and tilt

jmatthess, 2021-06-07:
Hi

Has anyone been able to determine if the library corrects for distortion as well as tilt? i'm trying to get the kinect skeleton into a 3D program and i cant tell if the lib corrects for lens distortion. I dont think it does for tilt but im not 100% sure.
If it does, has anyone been able to correctly undistort the 3d coordinate values of the joints?

im getting the skeleton position in this way and from what ive been able to see and from other issue posts, this is indeed outputting in meters.

floorClipPlane = _bodies.floor_clip_plane
floorW = floorClipPlane.w

joints = body.joints
skeleton = _kinect.body_joints_to_color_space(joints)
x = joints[PyKinectV2.JointType_SpineShoulder].Position.x
y = joints[PyKinectV2.JointType_SpineShoulder].Position.y + floorW
z = joints[PyKinectV2.JointType_SpineShoulder].Position.z

KonstantinosAng, 2021-06-07:
I think it corrects for lens distortion, but the tilt of Kinect position is not cosnidered a distortion. You can calculate the Kinect's tilt and Position from the floor using this code:

floor_vector = self._bodies.floor_clip_plane
kinect_distance_from_plane = floor_vector.w * 1000  # mm
kinect_tilt = np.arctan(floor_vector.z / floor_vector.y) * (180.0 / np.pi)  # degrees

Make sure that the kinect is tracking someone first!!!!!

jmatthess, 2021-06-07:
Thanks for the quick reply @KonstantinosAng learnt alot from reading your mapper library.
I dont suppose you would know how to apply the tilt angle to correct for the position data would you? ive tried looking at alot of resources but nit been able to find a methodology on how to correct for the tilt.

KonstantinosAng, 2021-06-07:
I always had the Kinect at a tilt anlgle but it did not produce any wrong data. Why do you think the position data need to be corrected ? Also keep in mind that the Kinect tracking algorithm works better between the range 0.5 to 4 meters.

jmatthess, 2021-06-07:
So im sending the xyz position data directly into a 3d program. i'm standing about 2 meters away from the kinect in a well lit room with no sun coming through the window to throw things off. and just by doing a TPose. From the foot to the head the Z offset gets pushed back a little bit more as you go up each joint. but also by tilting the camera i got slightly different height results.
i know that the kinect does have some margin for error for but the top of the skeleton appears to get pushed down a little with the camera tilted down a bit more. But if you're not experiencing the same, i may have to do a bit further testing as to what may be going on as the only other info im having to go off of are peoples post online and several people have mentioned compensating for tilt . this is one example, i cant find the original 
one that sent me down the assumption that tilt isnt compensated but this is one of many posts thats leading me to assume this https://stackoverflow.com/questions/57057802/kinect-v2-how-to-transform-kinect-v2-coordinate-to-real-life-measurements

KonstantinosAng, 2021-06-07:
Use this repo i created and see if you get distorted data in 3D. I had no problem when i created it and it could track me very accurately inside the room.

https://github.com/KonstantinosAng/PyKinect2-PyQtGraph-PointClouds