Kinect/PyKinect2 #98: black screen when opening

Chup123, 2021-12-07:
Hi,

When opening the example code after the modicications suggested (Update time.clock to time.perf_counter) i got the example running. The kinect turns on (lights are on) but nothing shows on the screen. Do you guys have any suggestions

KonstantinosAng, 2021-12-07:
try this https://github.com/Kinect/PyKinect2/issues/37 and also uninstall and install the Kinect Studio.

Chup123, 2021-12-07:
I tried that. It boots the kinect (all the lights turn on) and i see the pygame window. But after that nothing. And the thing is, in processing it works with the kinectPV2 library. I have an kinect model 1520. Do you know if that could be the problem?

KonstantinosAng, 2021-12-07:
Is it a Kinect for Windows or for Xbox ? Maybe the adaptor is wrong.

Chup123, 2021-12-07:
On the adapter it says kinect for windows. version 1637. Everything works, even the official c++ library, except for this library. And i want to do it in python, because my c++ isnt that great. It just seams that i dont get an frame. I work with python 3.9.7 on annaconda 32 bit.

KonstantinosAng, 2021-12-07:
check if it you can get an image with my repo:
https://github.com/KonstantinosAng/PyKinect2-PyQtGraph-PointClouds

Chup123, 2021-12-07:
sorry, does nothing

Chup123, 2021-12-07:
Fixed it. Create a conda virtenv with python 3.6.13 and installed everything. Replaced the files with the files in the repo and installed comtypes version 1.1.4 and it worked. Thanks for the help @KonstantinosAng

closed 2021-12-07 by Chup123