Kinect/PyKinect2 #99: Body is not being detected

EpicCodeWizard, 2022-01-02:
I ran `PyKinectBodyGame.py`, and stood in front of the sensor. No body is detected, yet when I use another library, my body is detected. Is there anything I can do?

KonstantinosAng, 2022-01-02:
what other library did you use ?

EpicCodeWizard, 2022-01-02:
Thanks for replying... LibKinect2 ([https://github.com/sshh12/LibKinect2](https://github.com/sshh12/LibKinect2)).

KonstantinosAng, 2022-01-02:
Try unistalling KinectStudio and installing it again. I remember that in order to detect the body you have to have a distance from the sensor.

nm9404, 2024-10-20:
Hey!!, I was going through the same situation and it got solved as soon as I downgraded python from ´3.8´ to ´3.6´ , and as stated in other issues make sure comtypes is installed in version ´comtypes==1.1.4´ 
Hope this helps! :)