Kinect/PyKinect2 #102: except COMError, err: SyntaxError: invalid syntax

Elisa-code184, 2022-06-22:
Hello everyone,
I've installed pykinect and comtypes == 1.1.4.
I've found the following issue running the PyKinectInfraRed.py example:

except COMError, err: SyntaxError: invalid syntax

I was wondering if anyone can help me.
Thank you.

zachvin, 2022-06-23:
I just had the same issue but in PyKinectBodyGame.py, and I'm pretty sure it has to do with the fact that you/I are using newer Python versions (I'm on 3.10.4) that has some syntax changes from earlier versions.

Add some parentheses so that the line looks like `except (COMError, err):`.
The same error will probably pop back up twice more, once for another COMError and one for a KeyError, so put those in parentheses too and I'm pretty sure the file will work after that.

Assuming you'll have the same issues I did, your next error will be something about the wrong comtypes version, I couldn't figure that one out so I just commented out line 2868 in PyKinectV2.py, which looked like this:
`from comtypes import _check_version; _check_version('')`
That's probably frowned upon but I did it anyway and my code connects to my Kinect so I'll just pretend nothing happened.

Further, you'll get an error about `time.clock()` not existing. Replace `time.clock()` with `time.perf_time()`, because time.clock() was [deprecated in Python 3.3 and removed in Python 3.8](https://github.com/wxWidgets/Phoenix/issues/1510#:~:text=The%20function%20time.,to%20have%20well%2Ddefined%20behavior.). 

I think those are all the fixes I did and my code works now, tomorrow I plan to aggregate all my changes/upload edited files to make the code work and post them somewhere so other people don't have to google everything.

Let me know if you have any other issues.

zachvin, 2022-06-23:
Did you check #37?
I got everything to work with 64-bit Anaconda v4.12.0 in a conda virtual environment with Python 3.6.13.
I just did all the instructions on the main github page, with these fixes:
- `pip install comtypes==1.1.4`
- after pip installing pykinect2, replace the .py files in the pykinect2 folder in Anaconda with the matching .py files from the github

I believe this is everything you need to do to get this code running on a modern Anaconda install.

Yudh0B, 2022-09-29:
> Did you check #37? I got everything to work with 64-bit Anaconda v4.12.0 in a conda virtual environment with Python 3.6.13. I just did all the instructions on the main github page, with these [...]

Traceback (most recent call last):
  File "c:\TA\PyKinect2-master\examples\PyKinectBodyGame.py", line 1, in <module>
    from pykinect2 import PyKinectV2
  File "C:\Users\grego\.conda\envs\TA\lib\site-packages\pykinect2\PyKinectV2.py", line 5, in <module>
    import comtypes
  File "C:\Users\grego\.conda\envs\TA\lib\site-packages\comtypes\__init__.py", line 381
    except COMError, err:

i've done all the required installation, still ended up with that same error, even tried reinstalling everything related to PyKinect2

Elisa-code184, 2022-09-29:
@Yudh0B I solved installing a less recent version of the Anaconda Environment and Python. 
If I am not wrong, I solved it with Python 3.7 and Anaconda 1.7.2

But anyway, try to downgrade the version until it works, I assume there is a problem in some library versions.
The other option is open and run the single code and solve the issues. I found some syntax errors in some modules because written in Python 2. I gave up at a certain point, and I picked the above solution.

Hope it could help.

sylent1951, 2024-06-04:
i think u need python3.6，i use 3.6.13 and after do all things i fix it